rweather/arduinolibs #6: Could you pack your different libraries in several repositories and push them to official Library Manager?

q2dg, 2015-10-11:
This would give a lot more visibility to your fantastic job.
Thanks.

P.D: See https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ for more information

harmon25, 2016-03-23:
I think the best way to keep these together would be in a new github organization. 

This repository would be divided into ~14 individual repositories within the new organization.

- docs (all doxy gen files)
- host (Compiling the Crypto library on a host system)
- One for each Library

This would also help organize any issues/questions reported for each library.

@rweather , if you are OK with this idea would you like to start the organization and invite me to it?

I can start splitting up the repositories, and setting them up to be published on the Arduino Library Manager, and PlatformIO for more people to discover!

rweather, 2016-03-23:
Hmmm.  I'm not sure whether I want to go that far yet.  There are dependencies between the libraries that would be difficult to manage without them coming as a package deal.

I have thought about splitting the crypto stuff out, but even that comes with dependencies.  It's not just Crypto, but TransistorNoiseSource, RingOscillatorNoiseSource, and in the future SecureShell (this one doesn't exist yet but I've been slowly working up towards SSH support).

Part of the problem is that the Arduino IDE is still dumb when it comes to libraries.  It thinks they are completely standalone.  It isn't possible to say "Anything that needs SecureShell must also link against Crypto" for example - all sub-dependencies must be listed explicitly as #include's in the sketch.  That's why the repo is effectively a dump of my entire sketchbook directory - it isn't easy to manage anything else.

The last I looked at the library properties stuff, dependencies were still not supported.

I'm open to suggestions.  I'll think about it, but it's a bit radical without an overall strategy to keep me/contributors sane.

harmon25, 2016-03-23:
You are right about the arduino library dependency problem, might not be worth figuring out.

Hopefully they add some dependency management functionality in the near future...

It would be doable for [PlatformIO](http://docs.platformio.org/en/latest/librarymanager/config.html) as it does provide a dependency option in the `library.json` file.

Would still need to split up the repositories, I think...

Alternatively, PlatformIO can be pointed to a download URL (.zip or .tar.gz).
A script could be conjured up that uses branches and releases to publish the code as PlatformIO libs...

heman22union, 2016-04-11:
@rweather what IDE do you use for Ardunio or compatible micro controllers? I understand that the Arduino IDE is not very good, but how are you getting this work work in its current structure?

rweather, 2016-04-12:
I use vim to edit the code, and the Arduino IDE to compile it and upload it to a device.  The repository is essentially a complete dump of my sketchbook directory so it already has the right layout to keep the IDE happy.  I do understand that this is pain when someone else wants to integrate it with their sketchbook directory.  My suggestion is to check out the code somewhere else and then copy the library or libraries of interest into your own sketchbook/libraries directory.

harmon25, 2016-04-22:
I have started giving this a go...
Created a fork where i will make branches for different libraries.
Starting first with [Crypto](https://github.com/harmon25/arduinolibs/tree/crypto-dep).
Moved the noise source libraries back to the root of Crypto, moved noise examples into Crypto examples folder.

Also created a [library.json file](https://github.com/harmon25/arduinolibs/blob/crypto-dep/library.json) for platform.io. I have not registered it though.

Can I do that @rweather or would you rather have control over the repos?
Updates will kind of be a pain this way...

rweather, 2018-04-07:
I have split out all of the non-crypto stuff into a separate repository:

https://github.com/rweather/arduino-projects

This repository has now been renamed the "Arduino Cryptography Library" and only contains cryptography-related stuff.

closed 2018-04-07 by rweather