rweather/arduinolibs #7: [libcrypto/speck] TestSpeck fails on Arduino Uno

ggramaize, 2016-01-20:
Building OS: Debian stretch
Arduino IDE version: 2:1.5.6.2+sdfsg2-3
Compiler version: gcc-avr 1:4.9.2+Atmel3.5.0-1 
Onboard bootloader version: unknown, will update this evening.

--------------------

Hi,

Both the full and low memory implementation of speck cipher don't match any of the reference test vectors on an Arduino Uno R3 target.

Moreover, the low memory implementation crashes when using a 192 bits or a 256 bits key size.

Here is the output I get when running the example sketch.

> State Sizes:
> Speck ... 275
> SpeckLowMemory ... 35
> 
> Speck Test Vectors:
> Speck-128-ECB Encryption ... Failed
> Speck-128-ECB Decryption ... Failed
> Speck-192-ECB Encryption ... Failed
> Speck-192-ECB Decryption ... Failed
> Speck-256-ECB Encryption ... Failed
> Speck-256-ECB Decryption ... Failed
> 
> SpeckLowMemory Test Vectors:
> Speck-128-ECB Encryption ... Failed
> Speck-192-ECB Encryption ...[random garbage printed here]

I'm expecting to receive an arduino mega next Friday. 
I'll try to reproduce the current issue on this target next weekend.

Kind regards

rweather, 2016-01-20:
Bummer.  Still works for me, but I have a Uno R1.

I sometimes get random garbage printed when the system stack overflows and
starts to intersect with data memory.  That causes a system crash and/or
other odd behaviours.

It's possible that is what is happening.  Try commenting out some of the
test vectors in TestSpeck.ino (e.g. the 128 and 192 bit ones) and see if
that helps recover enough data memory to make the remaining tests work.

In some of the other crypto tests, I have to force the test vectors off
into PROGMEM to get enough stack space to run the tests on a Uno.  I may
have to do that for Speck as well.

Thanks for the heads up.

Cheers,

Rhys.

ggramaize, 2016-01-20:
Problem solved using an up to date Arduino IDE version.
You should probably tell people to use your arduino IDE version or newer.

luz0928, 2017-05-25:
I'm still having this same issue and I have an up to date version of the arduino IDE.

MauroMombelli, 2017-05-28:
@luz0928 what version is it? did you download from arduino.cc or ardino.org?

rweather, 2017-05-28:
I just updated to arduino 1.8.2 and now I'm having issues. :-(

There seems to be several problems:

The new compilers generate larger data segments which then overlaps the stack region of memory and causes stack crashes.  This can be seen as a constant reboot loop.  I can probably reduce the data segment size by being more creative with PROGMEM declarations.

If I comment out much of the TestSpeck.ino code to reduce its size to stop the stack crash, then the test cases still don't pass.  But if I switch from the AVR assembly version to the plain C version, everything is fine.  So it looks like the newer compilers are doing something silly with the AVR assembly which causes the results to be wrong.

I suggest using the plain C version until I can figure out what the problem is.  In Speck.cpp, comment out the line that says USE_AVR_INLINE_ASM:

#if defined(__AVR__)
//#define USE_AVR_INLINE_ASM 1
#endif

rweather, 2017-06-03:
An update on this.  I've tracked the issue to the AVR assembly code in Speck::setKey().  The encryptBlock() and decryptBlock() functions work fine with AVR assembly.  I'm still not sure what part of setKey() the compiler is getting confused by though - I'm still investigating.

The latest Arduino IDE's use gcc 4.9.  The previous IDE's where the Crypto library worked used gcc 4.8.  Possibly something has changed in the handling of inline assembly between those two versions of gcc.

SpeckSmall and SpeckTiny have problems in encryptBlock() and decryptBlock() as well, but that is probably because they combine parts of setKey() with the block operations.  Once I can figure out what is wrong with Speck::setKey(), I should be able to fix the others.

luz0928, 2017-06-05:
I downloaded again the library, with the last commit and it passed the tests:

> State Sizes:
> Speck ... 275
> SpeckSmall ... 67
> SpeckTiny ... 35
> 
> Speck Test Vectors:
> Speck-128-ECB Encryption ... Passed
> Speck-128-ECB Decryption ... Passed
> Speck-192-ECB Encryption ... Passed
> Speck-192-ECB Decryption ... Passed
> Speck-256-ECB Encryption ... Passed
> Speck-256-ECB Decryption ... Passed

luz0928, 2017-08-15:
I'm experiencing this issue once again, I downloaded the library once more and now Speck does not pass any tests.

rweather, 2017-08-16:
This is a compiler issue that I still haven't figured out how to work around.  For now, if you are affected by it then you should disable AVR assembly in the Speck implementation as described above.

peter64, 2017-10-25:
I spent a bit of time comparing the byte code (as generated by avr-objdump.exe) between a working version and a non working version for the setKey routine based on the ASM version of the code. What I found is they are being compiled to exactly the same ASM byte code in the hex file (with the exception of one call command which references a different address). The call command itself is fine as they both properly reference the clean function.

As such I'm left thinking it must be something related to the state the MCU is configured in (or something outside of the setKey routine itself). Do you have any copies of the HEX for the setKey routine from a working version generated with gcc 4.8? Perhaps you can compare against the two versions I pasted below to see if you can spot any difference. Either that or maybe it's something to do with the byte offsets but I don't think so. I wish there was a debugger/emulator I could run it through to see where the data diverges.

Working then Failing (only difference)
`
    1cf6:	0e 94 af 0d 	call	0x1b5e	; 0x1b5e <_Z5cleanPvj>
vs
     f8e:	0e 94 e2 06 	call	0xdc4	; 0xdc4 <_Z5cleanPvj>
`

Working
`                      
2F923F924F925F926F927F928F929F92AF92BF92CF92DF92EF92FF920F931F93CF93DF93CDB7DEB7A7970FB6F894DEBF0FBECDBF3C014032510551F04831510551F04031510509F0A1C038E092E005C038E194E002C030E193E02EE1290FF301EE5EFE4F2083FA013897E60FF71FDE011196BFA3AEA302900D923A95E1F7F3013296FAA3E9A3FB01E40FF51F48E050E0A9A1BAA102900D924A95E1F7990F990F990F9850622E712CF3013197FDA3ECA3212CFD01AEA1BFA1392E412C512C6CA07DA0A20DB11DFD908D909D90AD90BD90CD90DD90ED9001911191219131914191519161917191800E911EA21EB31EC41ED51EE61EF71E842488E0280EA219B1099FE12922A30DB11D8D929D92AD92BD92CD92DD92ED92FD92380EA319B1093922000F111F221F331F441F551F661F771F011D000F111F221F331F441F551F661F771F011D000F111F221F331F441F551F661F771F011D082519252A253B254C255D256E257F250083118322833383448355836683778343946A9409F0A2CF60E270E0CD010E94AF0D81E001C080E0A7960FB6F894DEBF0FBECDBFDF91CF911F910F91FF90EF90DF90CF90BF90AF909F908F907F906F905F904F903F902F900895
`
Failing
`                   
2F923F924F925F926F927F928F929F92AF92BF92CF92DF92EF92FF920F931F93CF93DF93CDB7DEB7A7970FB6F894DEBF0FBECDBF3C014032510551F04831510551F04031510509F0A1C038E092E005C038E194E002C030E193E02EE1290FF301EE5EFE4F2083FA013897E60FF71FDE011196BFA3AEA302900D923A95E1F7F3013296FAA3E9A3FB01E40FF51F48E050E0A9A1BAA102900D924A95E1F7990F990F990F9850622E712CF3013197FDA3ECA3212CFD01AEA1BFA1392E412C512C6CA07DA0A20DB11DFD908D909D90AD90BD90CD90DD90ED9001911191219131914191519161917191800E911EA21EB31EC41ED51EE61EF71E842488E0280EA219B1099FE12922A30DB11D8D929D92AD92BD92CD92DD92ED92FD92380EA319B1093922000F111F221F331F441F551F661F771F011D000F111F221F331F441F551F661F771F011D000F111F221F331F441F551F661F771F011D082519252A253B254C255D256E257F250083118322833383448355836683778343946A9409F0A2CF60E270E0CD010E94E20681E001C080E0A7960FB6F894DEBF0FBECDBFDF91CF911F910F91FF90EF90DF90CF90BF90AF909F908F907F906F905F904F903F902F900895
`

rweather, 2017-10-31:
Just so everyone knows, I have a week off my day job this week, so I'm going to try finding this problem once and for all ... or die trying. :-)

The following is useful information: https://ucexperiment.wordpress.com/2012/02/21/arduino-assembly-language-listing-of-compiled-sketch-windows/

Summary: Use "avr-objdump -S" on the .elf file for the sketch to dump the assembly code with interspersed C++ code.  Much easier to spot differences between compiler versions that work and versions that don't.

rweather, 2017-11-01:
I have managed to fix the AVR assembly issues with the Speck class.  However, SpeckSmall and SpeckTiny still have issues - I'm working on them.

The new and old compilers were generating slightly different code, but the new code looked mathematically equivalent to the old.  So it is still a bit of a mystery.  The core issue seems to be that of register pressure - the newer compiler was finding it harder to find a free register to hold intermediate results so was spilling values to the stack more often.  Things may have gotten cross-wired because of the compiler's confusion.

I re-wrote Speck::setKey() so as to reduce the register pressure and now both the new and old compilers generate working code.

I'm now looking into how to apply the same trick to SpeckSmall and SpeckTiny.

rweather, 2017-11-03:
Fixed - I hope!  Speck, SpeckSmall, and SpeckTiny now all work with the old and new AVR compilers.  In the process I also improved performance slightly.

closed 2017-11-03 by rweather