rweather/arduinolibs #8: Clarification on Blake2s Implementation

harmon25, 2016-03-22:
Thanks for creating such awesome Arduino libs, this is a great resource! :+1: 

I have a question regarding HMAC digests on arduino vs other systems.

It is mentioned [here](https://rweather.github.io/arduinolibs/crypto.html) regarding this libraries BLAKE2 implementation:
> They are intended as high performance replacements for SHA256 and SHA512 for when speed is critical but exact bit-compatibility of hash values is not.

- I am attempting to create an HMAC on an Arduino or Raspberry Pi and verify the hash values on the other side. 
- It appears that the same input in the Blake2s C-ref produces a different hash than on Arduino using Blake2s HMAC from this Lib. 
- Is this what you are referencing in regards to exact-bit compatibility? 
 - An 8bit Arduino produces a different digest than the same input on a Raspberry Pi? 
- Any suggestions for best accomplishing this? 
 - I prefer blake2s due to variable digest size - am planning on transmitting payload + HMAC over [RF24Network](https://tmrh20.github.io/RF24Network/index.html)
 - payload size is limited(120 bytes), would prefer generating a small HMAC instead of truncating SHA256...

Any feedback is very much appreciated.
Thanks again for the awesome libraries!

rweather, 2016-03-22:
The "bit-compatibility" refers to SHA256.  BLAKE2s will produce different hashes than SHA256, of course.

Can you send me a link to the "BLAKE2s C-ref" library that you are using on the PI so that I can cross-check my implementation?  It should definitely be producing the same values as the one from https://blake2.net/.  If not, whoops!

Note also that BLAKE2 is not the same algorithm as the BLAKE submission to the SHA-3 contest.

It looks like BLAKE2 was standardised as RFC7693 late last year.  That RFC refers to a MAC code that is not the same as HMAC, as BLAKE2 has its own separate keyed hash mechanism (which I haven't implemented yet but probably should).  I'll cross-check my implementation against RFC7693 just to be sure.

harmon25, 2016-03-22:
I am using libb2 from the official BLAKE repo:  https://github.com/BLAKE2/libb2

Just a single call to the ['Simple API'](https://github.com/BLAKE2/libb2/blob/master/src/blake2.h) :
```
int blake2s( uint8_t *out, const void *in, const void *key, size_t outlen, size_t inlen, size_t keylen );
```

Here is some code, maybe I am doing something else wrong...

### Arduino Sketch
```
#include <Crypto.h>
#include <BLAKE2s.h>

#define HASH_LEN 16

BLAKE2s blake2s;

void setup() {
 Serial.begin(9600);
 
 char key[] = "testkey";
 char input[] = "{\"green\":0,\"red\":0}";
 
 uint8_t hashOfJSON[HASH_LEN];
 
 blake2s.resetHMAC(key, sizeof(key) );
 blake2s.update(input, sizeof(input) );
 blake2s.finalizeHMAC(key, sizeof(key), hashOfJSON, HASH_LEN);
 
 String hashStr;
 for(int i=0;i<HASH_LEN; i++){
  hashStr += String(+hashOfJSON[i], HEX);
 }
 
 Serial.println(hashStr);
}

void loop() {}
```
Arduino output: `89843bd440c471a3ddc3c14891579da4`

### C++ Program
```
#include <stdio.h>
#include <stdlib.h>

#include <blake2.h>

#define HASH_LEN 16

int main(int argc, char *argv[])
{
  uint8_t hashout[HASH_LEN];
  char key[] = "testkey";
  char input[] = "{\"green\":0,\"red\":0}";

  blake2s(hashout, input, key, HASH_LEN, sizeof(input), sizeof(key));

  for(int i=0; i<HASH_LEN; i++){
   std::cout << std::hex << +hashout[i];  
  }
  printf("\n");
  return 0;
}
```
C++ output : `d7e7708656491125250e2419d92a3`

Compiled with g++: ` g++ blaketest.cpp -lb2 -std=c++11 -Wall -o blaketest`

rweather, 2016-03-22:
Thanks for the example.  As I said, the BLAKE2 keyed hash mechanism is not the same as HMAC. So that would explain the discrepancy.  I'll try to find some time to put the BLAKE2 keyed hash into the Arduino library this weekend.

harmon25, 2016-03-22:
That is probably the issue...
Thanks a lot for taking a look!

You should add a donation button on the repos readme!
I am sure some people, myself included would be willing to give back for all the hard work you have put into this. 

#6 is also onto something, your libraries would be noticed/used more if they showed up in the Arduino IDE "Library Manager".  If you want some help with that, let me know.

rweather, 2016-03-23:
Done.  It was easier than I thought.  The test cases from RFC 7693 all pass, so it should be up to scratch now.  Give it a try.  There is a new override for the reset() function.  Use it like this:

blake2s.reset(key, keyLen, outputLen);
blake2s.update(data, len);
blake2s.finalize(hash, outputLen);

See here for the documentation: http://rweather.github.io/arduinolibs/classBLAKE2s.html#details

Let me know if it works end-to-end now.

If you want to tackle #6, then go for it.

harmon25, 2016-03-23:
That was it, thanks again for tackling this so quickly!

This has been resolved.

Will move discussion related to breaking up the libs into #6

closed 2016-03-23 by harmon25