rweather/arduinolibs #10: ED25519 - Arduino x Libsodium PHP

diegosilvars, 2016-11-09:
You can make a digital signature in PHP with libsodium and check with ED25519 library? I noticed difference in the private key size between the two libraries

rweather, 2016-11-09:
Ed25519 uses a 32-byte private key and a 32-byte public key.  Some API's concatenate these into a single 64-byte private key.  The Arduino library splits the 64-byte value into separate 32-byte buffers.  That may be the issue.

diegosilvars, 2016-11-09:
Many thanks for the reply.
If it is not too much to ask, could help me to sign a message in PHP and check the Arduino?

The Arduino receives the signature base 64 (do not know how to convert to uint8_t) and the public key would be fixed in the source code for the Arduino (do not know how to put on uint8_t).
Sorry for asking questions that can be considered basic, but am new to the Arduino.

My PHP code that will sign the message:

```
<?php

class AssinaIngresso {

public function assina(){
  
  $sign_secretkey = base64_decode("wLJKLltcIp3D95keAS8sQy4wl1j0jixi6BpEnU1T/+4mLtVYmq6lzYjSwaP8BsXhFVrm1UDo/fzzJ4AFk5HBMg==");
  $sign_publickey = base64_decode("Ji7VWJqupc2I0sGj/AbF4RVa5tVA6P388yeABZORwTI=");

    //Inicializa variável do hash de assinatura
    $signed_msg = "";
    $message = "This message";

    // Assinatura digital do ingresso
    $signed_msg = \Sodium\crypto_sign(
        $message,
        $sign_secretkey
    );

return base64_encode($signed_msg);
}
}
?>
```

And here is the Arduino code, I hope to verify the signed message in PHP.
Do not compile, publicKey can not receive decodedPublic because it is a variable of type char and so does the signature variable.
How do I convert the public key and the signature so that the verification work?

Please I need help.
Thank you.
```
#include <Base64.h>
#include <Crypto.h>
#include <Ed25519.h>
#include <RNG.h>
#include <utility/ProgMemUtil.h>
#include <string.h>

uint8_t privateKey[32];
uint8_t publicKey[32];
char* message;
size_t len;
uint8_t signature[64];

char* publicKeyReceived;
void setup() {

  Serial.begin(9600);

  publicKeyReceived = "Ji7VWJqupc2I0sGj/AbF4RVa5tVA6P388yeABZORwTI=";
  signature = "r4DG2+u9Gh71cK7waCheYTGu6vsqFYp5fKFZxcUZp/F81OkgF3XNr8N02ex4CA6KwFCnI4pvnl5cavuQUumjD1RoaXMgbWVzc2FnZQ==";

  int publicLen = sizeof(publicKeyReceived);
  int decodedPublicLen = base64_enc_len(publicLen);
  char decodedPublic[decodedPublicLen];
  base64_decode(decodedPublic, publicKeyReceived, publicLen);

//Publickey received decode but decodedPublic = Char
  publicKey = decodedPublic;
  
  message = "This message";
  len = sizeof(message);

  bool verified = Ed25519::verify(signature, publicKey, message, len);
  if (verified) {
      Serial.print("ok");
  } else {
      Serial.println("failed");
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

closed 2017-11-03 by rweather