rweather/arduinolibs #11: HKDF support

mlaz, 2016-11-10:
Hi!

It would be nice to have HMAC Key Derivation Function ( RFC5869: https://tools.ietf.org/html/rfc5869 )  support for Crypto's Hash classes. 
I wrote (and tested) an implementation which works with the SHA256 module and might also work with other Hash classes. If you are interested I would be glad to contribute to this project.  

Best regards,

Miguel A.

rweather, 2016-11-10:
Sure - that would be useful.  Easiest may be to make it a template like CTR/CTRCommon, so that the programmer can instantiate HKDF<SHA256>, HKDF<BLAKE2s>, etc.

mlaz, 2016-11-11:
This implementation consists only of 2 methods which do not depend on the Hash subclass, wouldn't it be easier just to add two methods to the Hash class? Unless you really want to make it a purely virtual class.

mlaz, 2016-11-13:
I just made a pull request with the implementation according with what you suggested. Thanks.

closed 2021-12-30 by rweather