rweather/arduinolibs #19: Error with Speck Library Example

luz0928, 2017-05-24:
I'm getting lots of compilation errors in the Speck example. I already updated my Arduino IDE.

The many errors include:

```
In file included from C:\Program Files (x86)\Arduino\libraries\Crypto\Ed25519.cpp:27:0:

C:\Program Files (x86)\Arduino\libraries\Crypto\utility/LimbUtil.h:60:48: warning: right shift count >= width of type

                         ((uint16_t)((value) >> 16))
C:\Program Files (x86)\Arduino\libraries\Crypto\utility/LimbUtil.h:61:25: note: in expansion of macro 'LIMB'

 #define LIMB_PAIR(x,y)  LIMB((x)), LIMB((y))

                         ^

C:\Program Files (x86)\Arduino\libraries\Crypto\Ed25519.cpp:100:5: note: in expansion of macro 'LIMB_PAIR'

     LIMB_PAIR(0x00000001, 0x00000000), LIMB_PAIR(0x00000000, 0x00000000),

```

And so on, many many errors like this.

I just opened the example and tried to compile it. What's going on?

rweather, 2017-05-24:
What is your Arduino board type?

Anyway, it should be fine, as those are warnings rather than errors.  The compiler is a little confused, but not fatally so.

luz0928, 2017-05-25:
I have an Arduino UNO R3. I will try to upload it to the board and see if the test vectors work.

luz0928, 2017-05-25:
Ok, I have uploaded successfully I think, but according to the serial monitor, all test vector encryption and decryption fail

rweather, 2017-05-28:
I have fixed the warnings in LimbUtil.h.  I will deal with the encryption/decryption test failures on the other ticket, https://github.com/rweather/arduinolibs/issues/7

closed 2017-05-28 by rweather