rweather/arduinolibs #20: TestCurve25519 is failing with a fresh download of the lib

Sossenbinder, 2017-05-31:
Hello,

I've been using several sections of this library without any problem, but I just wanted to try the Curve25519 part by going through the TestCurve25519.ino.

However, it seems to fail in every test case. I guess the problem must reside in the eval() function, since even dh1() is guaranteed to fail.

I did not make any changes to the file, and I have picked up the most recent version.

Is the Curve25519 known to be out of order currently, or am I doing something wrong?

rweather, 2017-05-31:
Which Arduino platform are you using?  There have been some issues with crypto algorithms that use AVR assembly with the most recent Arduino IDE's on Uno, Mega, etc.  See https://github.com/rweather/arduinolibs/issues/7 for example.  I'm trying to figure it out at the moment.

You could try going back to the plain C++ version by commenting out the following lines in Curve25519.cpp:

#if defined(__AVR__)
#define CURVE25519_ASM_AVR 1
#endif

Sossenbinder, 2017-06-01:
I am using an Arduino Yún.

Commenting out the mentioned lines really made everything work alright, the test cases are now running without any failure, thanks a lot for that already.

Is this workaround associated with any drawback?

rweather, 2017-06-01:
The plain C++ version is slower than the AVR assembler version, so there will be a slight performance penalty.  On the Uno, the AVR version of eval() takes about 2.7s compared to 3.1s for the C++ version.  The difference isn't huge but I'm still investigating, as potentially all of the primitives with AVR assembler versions are affected.

Sossenbinder, 2017-06-01:
Alright, thanks a lot for your help!

I don't think having slightly worse performance matters too much in my case, I'm just happy I found an implementation I actually got to work now :)

Feel free to keep me updated if you find out anything interesting.

freezedi, 2017-06-18:
Hello. Maybe you know whats wrong. 
Device: Wemos D1 mini (esp8266)
Curve25519.cpp
Line 345 
`result |= (uint8_t)((((uint16_t)0x0100) - check) >> 8);`

Com port show:
```
Fixed test vectors:
Computing Alice's public key ... ok (elapsed 294993 us)
Computing Bob's public key ... ok (elapsed 294844 us)
Computing Alice's shared secret ... ok (elapsed 295768 us)
Computing Bob's shared secret ... ok (elapsed 295702 us)

Diffie-Hellman key exchange:
Generate random k/f for Alice ... 
Exception (3):
epc1=0x4020217d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4022fae3 depc=0x00000000

ctx: cont 
sp: 3ffef560 end: 3ffef780 offset: 01a0

>>>stack>>>
3ffef700:  3ffe8568 00000000 3fff0b9c 3ffee598  
3ffef710:  fffffff8 3ffee598 3ffee598 3ffee578  
3ffef720:  3fffdad0 001acd1b 3ffee72c 40201f54  
3ffef730:  3ffe86f0 0014dfb2 3ffee5b8 40202bc6  
3ffef740:  3fffdad0 3ffe8638 3ffee72c 3ffee750  
3ffef750:  3fffdad0 00000000 3ffee72c 402020c3  
3ffef760:  feefeffe feefeffe 3ffee748 402031a4  
3ffef770:  feefeffe feefeffe 3ffee760 40100114  
<<<stack<<<
```

rweather, 2017-06-18:
result |= (uint8_t)((((uint16_t)0x0100) - check) >> 8);

That line is in a function that uses PROGMEM declarations.  There have been issues with Crypto/utility/ProgMemUtil.h on ESP8266 in the past and some contributed patches have been applied.  It is possible that there are still problems with it.  Unfortunately I don't have access to an ESP8266 to investigate this myself.

rweather, 2017-11-03:
I have disabled AVR speed-ups in Curve25519 in the main repository for now.  I will try to figure out what the problem is when I have some time.

xpanis, 2018-02-17:
Hello, I work with wemos d1 too, and I wanna use curve25519.h (DFH algo), but have the same problem, will you fixed it in near future? Thank you so much.

freezedi, 2018-02-17:
Maybe it'll help to you.
I changed in Curve25519.cpp
**line 344:**
uint8_t check = (pgm_read_byte(point + 31) ^ k[31]) & 0x7F;
**on:**
uint8_t check = (pgm_read_byte(&(points[0]) + 31) ^ k[31]) & 0x7F;

and line 347 the same 
check |= (pgm_read_byte(point + index - 1) ^ k[index - 1]);
check |= (pgm_read_byte(&(points[0]) + index - 1) ^ k[index - 1]);

wemos seems point as array, you can use the address of value. GL.

xpanis, 2018-02-17:
No way, you are my new god :* - thank you s much. It works. Yes it makes sense, but can you explain me, how you get that problem is right there? -> I mean some track method (for some similar future problems). Are you make some breakpoints/prints into curve25519.cpp file?

MauroMombelli, 2018-02-17:
it seems to be a compiler issue, maybe with optimizing where he should not,
by playing around like this the compiler take a different "execution path"
and the bug is eliminated.
just my 2 cent

rweather, 2018-02-17:
Thanks for figuring out what was causing the failure on esp8266.  I'll make the change to the main repository.  And yes, technically the two should be identical but obviously the compiler is handling array accesses differently than pointer accesses for pgm.

rweather, 2018-02-17:
Change made.  Instead of "&(points[0]) + 31", I used "&(points[31])".  Let me know if that slight different still causes problems and I'll revert to "&(points[0]) + 31".

rweather, 2018-04-02:
I have fixed some issues with ESP8266 and ESP32 modules that are related to the problems described on this ticket.  Details here:

https://github.com/rweather/arduinolibs/issues/28

I'm keeping this ticket open for now as I still need to figure out why the AVR speed-ups for Curve25519 stopped working.