rweather/arduinolibs #21: Ed25519 is using suspiciously high amounts of program memory

Sossenbinder, 2017-06-09:
Hello!

I have used your library to quite some extent, and so far I never had any problems with it. However, I tried to use the Ed25519 today, and I instantly ran out of memory on my Arduino Yún.

Before using Ed25519, I was using around 15kB program memory, while comfortable having Curve25519 and ChaChaPoly implemented already. However, simply including Ed25519.h, and calling sign() and verify() is shooting my usage to a little bit over 30kB. This was quite surprising, considering how lightweight all the other parts are.

I've been going through the Ed25519 code, and I couldn't really determine what was causing this huge hardware requirement.

Is this the usual behaviour/usage?

rweather, 2017-06-09:
Yeah, it's bigger.  It's not just the code in Ed25519.cpp, but it also links in extra helper functions from Curve25519.cpp and BigNumberUtil.cpp that Curve25519 itself doesn't use.  With Curve25519 alone, the linker can remove a lot of those functions from the final size.

closed 2017-06-09 by rweather

sairathb, 2020-04-10:
is there a way to clear the memory usage for ED25519?

rweather, 2020-04-11:
Flash program memory is fixed at compilation time, so other than re-flashing the device with a completely different program, that memory cannot be reclaimed.

ESP8266 also has lots of RAM, but apparently only 4K of it is devoted to stack space in the default Arduino configuration.  Ed25519 needs quite a bit of stack space, and if other components of your sketch also need lots of stack space, it can get a little squeezy.  Moving some parts off into global memory and the heap can help.

There are some hints here as to how to increase the default stack size on ESP8266: https://github.com/esp8266/Arduino/issues/2919