rweather/arduinolibs #26: Adding Padding using Crypto library

fralomb, 2017-11-26:
Looking through the Crypto Library i wasn't able to found an implementation of the different types of standard paddings to be added in the case the data size to be cipher/decipher is not a multiple of the base block. Why this feature is not present?

prochat, 2017-11-26:
Because it’s already made … you can change the #define if you want to do it by yourself

… or I’m not sure I get your point.

Ph.R.

fralomb, 2017-11-26:
@prochat probably i did not explain my doubt very well.. i cannot find a mechanism in the library which add a specified padding to a block of data that isn't a multiple of the block size of a specified algorithm. Is there a way in the library to accomplish this or i have to specify this by myself? In the first case, can explain me how i can do it? Thanks.

rweather, 2017-11-26:
Hi there.  The hash algorithms have in-built padding support compatible with their respective standards.  Nothing special needs to be done for them.

I assume that you are referring to padding for block ciphers in CBC mode.  There is no explicit support for padding in the Crypto library because there are many different padding schemes - pad with zeroes, pad with a byte set to N if there are N padding bytes, pad with a 1 bit followed by a zero bit and finally a 1 bit.  Which one?

Also, dealing with the last block in CBC mode is a pain in the neck for both encryption and decryption, as an extra block needs to be added if the plaintext was block-aligned to begin with.  So it was easier to say "the application pads the data itself and then passes the block-aligned data into the CBC mode".

In any case, CBC mode is not recommended for new designs as it is difficult to make the padding have constant-time behaviour.  It is better to use CTR mode which doesn't require padding.  Or better yet, EAX mode for encryption and authentication (AEAD).

If you do still want to use CBC mode, then you'll have to pad the block yourself before passing it to the library for encryption, and remove the padding yourself after decryption.

closed 2017-11-26 by rweather