rweather/arduinolibs #28: Exception (3) problem with Wemos D1 - methods: update/finalize

xpanis, 2018-03-16:
Hello there,

I have some memmory problem (on Wemos D1... on Arduino UNO works it good) with methods: update/finalize in lib: blake2s.cpp (Reset method work properly on both hw).

When I call this ones (e.g.: `blake.update(salted_code, sizeof(salted_code));`), in console I can see only:
```
Exception (3):
epc1=0x4020211d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40230420 depc=0x00000000

ctx: cont 
sp: 3ffef600 end: 3ffef920 offset: 01a0

>>>stack>>>
3ffef7a0:  feefeffe feefeffe feefeffe 3ffee8c8  
3ffef7b0:  6b08c647 bb67ae85 3c6ef372 a54ff53a  
3ffef7c0:  510e527f 9b05688c 1f83d9ab 5be0cd19  
3ffef7d0:  6a09e667 bb67ae85 3c6ef372 a54ff53a  
3ffef7e0:  510e523f 9b05688c 1f83d9ab 40100574  
3ffef7f0:  5be0cd19 3c6ef372 a54ff53a 1f83d9ab  
3ffef800:  a54ff53a 1f83d9ab 3ffe84c1 40203fb8  
3ffef810:  00000009 00000019 3ffee8c8 3ffee8ec  
3ffef820:  3ffe84c0 3ffee6f0 3ffee8c8 3ffee6f0  
3ffef830:  00000002 3ffee6f0 3ffef860 402024ef  
3ffef840:  feefeffe feefeffe 3ffee8c8 3ffee8ec  
3ffef850:  3ffe84c0 3ffee6f0 3ffee8c8 40201ce1  
3ffef860:  3ffe85f8 feefeffe 6b08c647 bb67ae85  
3ffef870:  3c6ef372 a54ff53a 510e527f 9b05688c  
3ffef880:  1f83d9ab 5be0cd19 00000000 00000000  
3ffef890:  00000000 00000000 00000000 00000000  
3ffef8a0:  00000000 00000000 00000000 00000000  
3ffef8b0:  00000000 00000000 00000000 00000000  
3ffef8c0:  00000000 00000000 00000040 00000000  
3ffef8d0:  feefef40 feefeffe feefeffe feefeffe  
3ffef8e0:  feefeffe feefeffe feefeffe feefeffe  
3ffef8f0:  feefeffe feefeffe feefeffe 3ffee8ec  
3ffef900:  3fffdad0 00000000 3ffee8e4 40203d54  
3ffef910:  feefeffe feefeffe 3ffee900 40100114  
<<<stack<<<
```

Firstly I think it is similar problem, resolved here: 
[https://github.com/rweather/arduinolibs/issues/20#issuecomment-366459155](url)

But I don't know how resolved it, or I can print some help notes via library methods).
I add @freezedi, @jakubfindura cause he has similar problem, and resolved it.

I hope that somebody of you help me with this problem.

Best regards
Ján

xpanis, 2018-03-17:
Hey again,

I'll try to know what is it problem, and I got:
Problem is with function: processChunk();

Right there:
`for (index = 0; index < 8; ++index)	{
        state.h[index] ^= (v[index] ^ v[index + 8]);`

When I assign (or xor) variable into `state.h[index];` it crashed.
But when I assign e.g. into `state.h[index] = 0xFFFFFFFF;` it will works.
When I do: `uint32_t new_index = (v[index] ^ v[index + 8]);` it works, but next if I'd like to assign (or xor) into `new_index = (v[index] ^ v[index + 8]);` it fail again :/

Thank you for any help.

Best Regards
Ján

rweather, 2018-03-31:
I borrowed a WeMos D1 Mini from my brother.  The BLAKE2s test case and others work for me with the following configuration:

- Arduino IDE 1.8.2
- 2.4.1 version of the ESP8266 board support code from https://github.com/esp8266/Arduino
- Tools -> Board configured as "WeMos D1 R2 & mini".

Maybe whatever compiler issue was occurring before has been fixed now?  Or it is even more subtle.

I have noticed some issues with other test programs like TestAES which run the tests for a longer amount of time.  If the code spends more than 5 seconds in the setup() or loop() function without returning to the caller, then a Soft WDT Reset exception will occur.  This can be avoided by calling wdt_disable() at the start of the setup() function.  In production code it may be necessary to call wdt_reset() periodically to feed the watchdog if the program spends a lot of time in setup() or loop().

I'm still working through the other Crypto examples to make sure they all work in an ESP8266 configuration with the latest toolchain.  And there may be need to be some other changes for EEPROM access and chip ID's in the RNG class.

rweather, 2018-04-02:
I have made some modifications to the library and all crypto examples now work with my brother's ESP8266 WeMos D1 Mini and ESP32 Dev Module.  For the ESP8266 it was mostly watchdog issues, with a few other things here and there.  I have collected up some recommendations for people who are using ESP8266 here:

http://rweather.github.io/arduinolibs/crypto_esp.html

I also modified the RNG class to use the hardware TRNG's on both ESP8266 and ESP32.

Hopefully this will put the ESP-related issues to rest now.

closed 2018-04-02 by rweather

xpanis, 2018-04-16:
Hello Rwather, I just wanna inform you, that I instal the newest version of IDE + updated library + ESP8266 core as you mentioned, and all work very well. Thank you again for you're response and have a nice day :) Bye

PS: If I'll find some problem thing in library, I will inform you, to improve user experiance in your lib.

rweather, 2018-04-16:
Fantastic - I'm glad it is working for you now.  I look forward to any comments as to how to improve the library.