rweather/arduinolibs #31: How to prevent the crypto library from using the watchdog timer?

bmanga, 2018-06-07:
CRYPTO/RNG.cpp uses the watchdog to generate entropy, however I need to use it for other purposes.
I see that part is guarded by a macro, but setting RNG_WATCHDOG to 0 causes compilation errors.
Is there a simple way to disable it?

rweather, 2018-06-07:
Try commenting the line out completely rather than setting it to 0; i.e.:

//#define RNG_WATCHDOG 1

You may also need to comment out the "no hardware random number source" error/warning.  You will need a separate noise source to replace the watchdog of course.

I hope this helps.

bmanga, 2018-06-07:
Yes, it fixes is thanks!

closed 2018-06-07 by bmanga

ortegafernando, 2020-03-13:
Hi, I am using in the same sketch LowPower library and this library. I have problems to use another source of noise. 
Could I get any simple example ?
Thanks.