rweather/arduinolibs #34: Compilling for ESP32

Macrosii, 2018-11-11:
Hi, thanks for this great library.
I am using the Crypto library and AES examples.
I can compile and run ok on Teensy, M0 and M4.  However when trying to compile for ESP32 I get the following error.  I am using Arduino IDE 1.6.7 and latest version of this library.  
TestAESCBC:172: error: 'AES128' does not name a type

 AES128 aes128;

 ^

TestAESCBC:173: error: 'AES192' does not name a type

 AES192 aes192;

 ^

TestAESCBC:174: error: 'AES256' does not name a type

 AES256 aes256;

 ^

rweather, 2018-11-11:
Weird.  I just tried to recompile it with my setup: Arduino 1.8.2 and the "ESP32 Dev Module" board type.  It seems to compile OK, so maybe it is something to do with the older 1.6.7 IDE?

The ESP32 version of the AES code tries to use the hardware cryptography module inside the ESP32.  You can temporarily disable it by changing the following lines at the top of AES.h:

#if defined(ESP32)
#define CRYPTO_AES_ESP32 1
#else
#define CRYPTO_AES_DEFAULT 1
#endif

Change them to simply:

#define CRYPTO_AES_DEFAULT 1

I hope this helps.

Macrosii, 2018-11-12:
Thanks very much for such a prompt reply.  I installed IDE 1.8.7 fresh on my laptop, the ESP32 core from board manager any your library under usr/documennts/arduino/libraries/Crypto/.
OS is Windows 7 Premium

I get the same error with #define CRYPTO_AES_DEFAULT 1 or #define CRYPTO_AES_ESP32 1.
I renamed the ESP aes.h in hwcrpto and it complains its missing with CRYPTO_AES_ESP32 1, as expected.
There must be something strange with my ESP32 setup or install.  It compiles with all the other boards I have tried.  I will try asking on the Arduino forums.  So far searching on  error: 'XXXX' does not name a type seems to be missing files, but that is not the case here.  If i find a solution i will post back.  thanks

Macrosii, 2018-11-12:
Should I post the debug output here ? its 152 lines ?

Macrosii, 2018-11-12:
[aes_compile error.zip](https://github.com/rweather/arduinolibs/files/2571057/aes_compile.error.zip)

Compile output - verbose.

Macrosii, 2018-11-12:
Yes it just your AES example slightly modified where I wanted to see what happens to the decrypted text if you change one byte of the cypher text.
[TestAESsteal.zip](https://github.com/rweather/arduinolibs/files/2571680/TestAESsteal.zip)
. first 3 lines are 

#include <Crypto.h>
#include <AES.h>
#include <string.h>

It complies for most board tyes, just the ESP32 boards give the error.

Macrosii, 2018-11-14:
FYI
To rule out a Windows 7 issue I installed fresh on my wife's Windows 10 machine.
IDE 1.8.7
ESP32 core 1.0.0
Library installed to /usr/documents/arduino/libraries/Crypto
Compile example /Crypto/examples/TestAES/TestAES.ino

Same issue, compiles ok for other board types but gives the same error above for ESP32.
If I find a solution I will post back.
Regards

LQ1234, 2020-03-22:
I believe this is the same issue I am having with my arduino nano 33 ble. In both, crypto.h and aes.h are already defined:
NANO:
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/cores/arduino/mbed/features/mbedtls/mbed-crypto/inc/mbedtls/aes.h
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/cores/arduino/mbed/features/mbedtls/mbed-crypto/inc/psa/crypto.h

ESP:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/wpa_supplicant/crypto/crypto.h
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/rom/aes.h

Not sure what to do about it tho

LQ1234, 2020-03-22:
I put the lib into src/ and imported it like this:
#include "src/Crypto/Crypto.h"
#include "src/Crypto/AES.h"
You guys have a easier api tho

rweather, 2020-03-23:
I may have to consider renaming the class in my library since so many other libraries seem to be using "AES128" and "AES256" now too.  I was hoping not to have to do that. :-(

ortegafernando, 2021-01-04:
Hi, I have the same problem. how do I have to solve it? Thanks. (esp32)

rweather, 2022-01-15:
I have added some macros to #define my class names to something else to try to work around the system headers.

The long-term solution is for esp-idf to rename their enumerations to ESP_AES128, etc so that the system headers don't pollute the global namespace with generic names.