rweather/arduinolibs #35: Using GCM<AES256> in another library

luckwaski, 2018-11-27:
Hello,
I'm having a hard time running GCM<AES256> from within another library on Arduino.

Heres my sketch:

`#include <avr/wdt.h>
#include <Crypto.h>
#include <AES.h>
#include <GCM.h>

#include <Cryptotest.h>

Cryptotest cryptotest;

void setup() {
  wdt_disable();
  
  Serial.begin(250000);
  delay(500);
  Serial.print("START");
  cryptotest.test();
}

void loop() {

}`

And the test library:

`
#ifndef Cryptotest_h
#define Cryptotest_h

class Cryptotest
{
  public:
    Cryptotest();
    void test();
    GCM<AES256>*gcmaes256=0;
};

#endif`

`
#include "Arduino.h"
#include <SPI.h>
#include <Crypto.h>
#include <AES.h>
#include <GCM.h>
#include "Cryptotest.h"

Cryptotest::Cryptotest()
{
}

void Cryptotest::test()
{
    gcmaes256->clear();
}
`

What happens is atmega is going for an infite restart loop "START" displaying over and over in the serial.
Any ideas would be appreciated.

Thank you

rweather, 2018-11-27:
It looks like the gcmaes256 pointer was not initialised and is still null.  So there is no object there and the program crashes causing a system restart.  You could try changing the Cryptotest constructor as follows:

Cryptotest::Cryptotest()
{
    gcmaes256 = new GCM<AES256>();
}

You'll also need to add a destructor to clear the object up:

Cryptotest::Cryptotest()
{
    delete gcmaes256;
}

I hope this helps.

luckwaski, 2018-11-27:
Thank you! It helped - I just changed the initializer a bit:

gcmaes256 = new GCM<AES256>();