rweather/arduinolibs #36: Test Ed25519 "Crypto" library is not giving timing response for Signature verification

HEMANT8712, 2019-05-17:
Hello @rweather,

I am trying to execute Signature-verification using ED25519 Curve using text example given in “Crypto” library. I was unable to get timings for signature verification.

May you please suggest any solution, 
[image]

rweather, 2019-05-17:
Ed25519 needs quite a bit of RAM so if you are running it on an Arduino Uno or something similar with a small amount of memory, sometimes it can run out and crash.

I believe I may have derived some of the timings on my Web site using an Arduino Mega to get enough RAM for the operation to work.  Even if the original timings were done on an Uno, then changes to the Arduino compiler over the years may have altered the memory layout enough that it doesn't fit any more.

HEMANT8712, 2019-05-17:
Thanks, @rweather, Is there a way to test this on Arduino Uno? as I am doing my project based on low-level devices (8-bit microcontroller) with less memory and Arduino Uno is of the correct specification.   
May you please suggest.

rweather, 2019-05-17:
It's not easy - the amount of memory required for signature verification is pretty severe so you need to watch every byte used everywhere in the program.  Keep things off the stack as much as possible.  It only just barely fit in the first place.

The sign() function uses less memory, so you could look at the larger system design for that.  If the Uno is communicating with a desktop machine, then try to arrange your high-level protocol so that the Uno does the signing and the desktop does the verifying.  Then you don't need to worry about the memory requirements of verify().

Or have one Uno sign and the other verify - the TestEd25519 example has everything in one program but if you can strip away unnecessary parts you may be able to reclaim a little memory for the verifying device.

Otherwise you might need to go into the code and rearrange how it works.  If you look at the verify() function, I reuse the memory inside the "hash" object for the "k" buffer.  There may be other places where buffers could be overlapped, or the steps can be done in a different order that allows memory to be reclaimed.

HEMANT8712, 2019-05-28:
Thanks, @rweather, Is there a way to test this on Arduino Uno? I have tried it on Arduino Mega2560, it is working fine and tested only verification process on Arduino Uno, it is also working.

I have one query, I am doing mathematical analysis (like the number of point multiplication and addition used) also, may you please let me know which standard algorithms you have used and where I can found their mathematical explanation.

rweather, 2019-05-28:
The main source of information on Ed25519 is from the RFC specification:

https://tools.ietf.org/html/rfc8032

The "Informative References" section contains links for the underlying academic papers that explain the mathematics involved.

HEMANT8712, 2019-05-29:
Thanks, @rweather, for the help.

closed 2019-06-14 by HEMANT8712