rweather/arduinolibs #37: Timing difference between AES encryption and decryption

HEMANT8712, 2019-05-29:
Hello @rweather ,

I have one final query. I am checking the time for encryption and decryption for AES. and it's nearly double. But when I trying to look for the standard you used to build that code, it's only showing Wikipedia link. 
I need to ask from which standard you design the AES algorithms for Crypto library.  Please suggest as we need to check the mathematical reasoning for the time difference in encryption and decryption.

rweather, 2019-05-29:
The Wikipedia description is sufficient to implement it, but there is also the NIST specification which should be linked somewhere in the References.

The key performance difference between encryption and decryption is in the MixColumns step where it multiplies the state by a matrix of constants.  The inverse of this matrix is more complex so it takes longer to perform the step in reverse for decryption.  See https://en.wikipedia.org/wiki/Rijndael_MixColumns

On 32-bit systems it is possible to combine the SubBytes, ShiftRows, and MixColumns/InverseMixColumns steps into lookup tables.  This makes the speed about the same for both directions.  But 8-bit platforms don't have enough memory for that so I didn't bother doing that.

In practice the slowness of decryption isn't an issue because you should be using a mode like CTR or GCM that doesn't require the block decrypt operation.  Only CBC requires decryption and you shouldn't use that due to timing attacks against the padding algorithms.

closed 2019-05-30 by rweather

HEMANT8712, 2019-05-31:
Thanks Sir,

For the help and Explanation.

Regards,
Hemant