rweather/arduinolibs #40: Using Crypto and CryptoLW library with Nodejs

HEMANT8712, 2019-07-08:
Hello @rweather ,

We are trying to use Ed25519, Curve25519 functions with javascript module. But there are many .cpp files linked with each other. 

May you please suggest how to convert Crypto folder into the library, such that we can use it with my local javascript function.

Regards,
Hemant

rweather, 2019-07-10:
The host/Crypto directory contains a plain Makefile that builds the whole thing as a library under Linux systems for running tests on a desktop system.

The main core for the curve functions is in Curve25519.cpp, Ed25519.cpp, BigNumberUtil.cpp, and SHA512.cpp.  Copy those out and any other files that they try to build against and you should have a good start.

Curve25519 and Ed25519 do depend upon RNG.rand() but you can replace that with a host version instead like host/Crypto/RNG_host.cpp to avoid dragging in lots of other dependencies.

HEMANT8712, 2019-07-11:
Hello @rweather 
 
Thanks for the reply. I am sorry but I got confused with a sentence, regarding Curve25519.cpp, Ed25519.cpp.
Functions of these two important files I need to use through the library.  **What is meant by "Copy those out and any other files that they try to build against and you should have a good start.".**
 And on using make file, will it contains their functions.
Please suggest.

Thanks and Regards,
Hemant