rweather/arduinolibs #45: Decrypted data differ from origin data

zhekaus, 2019-11-17:
I am trying Acorn128. My trial code is very simple:
```
	const byte len{16};

	byte data[len]{1, 2, 3};
	byte cipher[len]{};
	byte decrypted[len]{};

	acorn.encrypt(cipher, data, len);
	acorn.decrypt(decrypted, cipher, len);
```
And I get wrong decrypted data. Here is the output:
```
data:
1:2:3:0:0:0:0:0:0:0:0:0:0:0:0:0:
encrypted:
19:CE:6F:7E:96:16:89:6A:39:FE:8:70:E2:4:95:9E:
decrypted:
DF:29:FC:C5:19:7:D5:2:58:35:16:B8:61:AA:1B:D6:
```

rweather, 2019-11-17:
You need to re-initialise the cipher object before decryption.  Acorn128 keeps track of the current stream position.  The encryption request is for bytes 0 to 15 and the decryption request is for bytes 16 to 31.  Because the offsets are different they will encrypt/decrypt differently.

You should call setKey(), setIV(), and optionally addAuthData() again before decryption to reset the stream offset to 0.  See the TestAcorn example code for the pattern.

zhekaus, 2019-11-17:
Oh, thanks a lot!

I was thinking that IV is optional.

I was experimenting with this library, and I found that length of IV must be exact 16. Otherwise decryption fails.