rweather/arduinolibs #53: Memory reset in ESP8266 on signing with ED25519

sairathb, 2020-04-10:
Hi @rweather, I really appreciate the help you provide with this library. Its very comprehensive and I want to thank you for the help with AES256. Now I am trying to perform a asymmetric encryption using ED25519 curve. Here is the code I wrote following your example:
```
  long _millis = millis();
  String millis = String(_millis, DEC);
  Serial.println(millis);
  uint8_t privateKey[32];
  uint8_t publicKey[32];
  Ed25519::generatePrivateKey(privateKey);
  Ed25519::derivePublicKey(publicKey, privateKey);
  dumpByteArray(privateKey, 32);
  dumpByteArray(publicKey, 32);
  /*byte signature[64];
  Ed25519::sign(signature, privateKey, publicKey, (byte*)millis.c_str(), millis.length());
  dumpByteArray(signature, 64);
  int encLen = base64.encodedLength(64);
  char verifier[encLen];
  base64.encode(verifier, (char*)signature, 64);
  Serial.println("Encoded signature");*/
```
I initially tried to use a key pair that I generated using PyNaCl library. But that was throwing the memory dump on `Ed25519::sign`, then I though of generating the keys and dumping it, as you see above. I am still getting the memory corruption issue. Can you please guide me in this. 

I am using a NodeMcu ESP8266 board.

sairathb, 2020-04-10:
I got it fixed, I declared all the objects within the function that was going beyond the available space. I moved the classes that is required for all function globally and it is working now.

closed 2020-04-10 by sairathb

rweather, 2020-04-10:
OK - thanks for letting me know.  It will come in handy in the future if other people have a similar problem.