rweather/arduinolibs #57: Possible to add modular exponentiation to BigNumberUtil?

MarcelGerber, 2020-06-27:
Hi, I really quite enjoy working with your library, great work there!

Would it be possible to also implement efficient modular exponentiation (i.e. `modPow(b, e, modulus) = (b ** e) % modulus`) in BigNumberUtil? That would be very helpful for my application where I need to implement RSA.
If modular exponentiation is already possible in any way, a pointer in the right direction would also be helpful :)

rweather, 2020-06-28:
Unfortunately I don't have modexp at the moment as implementing general-purpose big number division with remainder is a challenge.  Elliptic curves like Curve25519 can do division more efficiently because of special-case divisors like 2^255 - 19.  So it wasn't a high priority to handle the general case.

Some more recent Arduino modules like the ESP32 have built-in RSA acceleration, so you might be able to make use of that.

MarcelGerber, 2020-06-28:
Thank you for your response. Yes, I absolutely realize that implementing such a feature comes with its pitfalls and is not easy.

For anyone else looking for a similar feature, [https://github.com/wolfSSL/wolfssl](wolfSSL) has an [https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/integer.h](integer.h) file which provides all the big-integer-general-purpose operations you could ever wish for.
It needs some getting-used-to, some configuring to run on an ESP8266, and the API is less on point than the one provided by this library here, but nonetheless I was able to implement the Rabin cryptosystem using it.