rweather/arduinolibs #58: AES128 - CTR Decrypted data is different from original

franciscone14, 2020-06-30:
I'm learning cryptography and I manage to use AES without the CTR, but when I added the CTR mode the decrypted data does not match the original. In my application I'll use an Arduino UNO r3 to send a luminosity sensor data for a node-red applications using MQTT and in the Json format (the application will get more complex and I plan on using the node-red interface to open a door, for instance). Sorry for the newbie doubt, but can you help me to understand what is wrong with my code?

`#include <SPI.h>
#include <Ethernet.h>
#include <PubSubClient.h>
#include <Crypto.h>
#include <AES.h>
#include <string.h>
#include <CTR.h>

CTR<AES128> ctraes128;

byte payload_encrypted[16];
byte payload_decrypted[16];

byte iv[16] = {
    0x00, 0x00, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01
};
const char key[16] = "5TGB&YHN7UJM(IK<";
const char text[16] = "Single block msg";

bool prepareCipher(Cipher *cipher, byte key[], byte iv[]){
  cipher->clear();
  if (!cipher->setKey(key, cipher->keySize()))
  {
    return false;
  }
  if (!cipher->setIV(iv, cipher->ivSize()))
  {
    return false;
  }
  return true;
}

void encryptData(Cipher *cipher, const uint8_t data[], uint8_t encryptedData[], size_t tam){
  size_t posn, len, inc;
  inc = 1; // idk what it does but it was used in the example
  for (posn = 0; posn < tam; posn += inc)
  {
    len = tam - posn;
    if (len > inc)
      len = inc;
    cipher->encrypt(encryptedData + posn, data + posn, len);
  }
  cipher->clear();

  Serial.print("Payload Encrypted: ");
    for(int i = 0; i < sizeof(text); i++){
        Serial.print(payload_encrypted[i]);
        Serial.print(":");
    }
    Serial.println();
}

void decryptData(Cipher *cipher, const uint8_t encryptedData[], uint8_t data[], size_t tam){
  size_t posn, len, inc;
  inc = 1; // idk what it does but it was used in the example
  for (posn = 0; posn < tam; posn += inc)
  {
    len = tam - posn;
    if (len > inc)
      len = inc;
    cipher->decrypt(data + posn, encryptedData + posn, len);
  }
  cipher->clear();

  Serial.print("Payload Decrypted: ");
    for(unsigned int i = 0; i < sizeof(text); i++){
        Serial.print(payload_decrypted[i]);
        Serial.print(":");
    }
    Serial.println();
}

void setup()
{
    Serial.begin(9600);
    Serial.print("Original text msg: ");
    for(unsigned int i = 0; i < sizeof(text); i++){
        Serial.print(text[i], HEX);
        Serial.print(":");
    }
    Serial.println();

    prepareCipher(&ctraes128, (byte*) key, iv);
    encryptData(&ctraes128, (byte*)text, payload_encrypted, sizeof(text));
    decryptData(&ctraes128, payload_encrypted, payload_decrypted, sizeof(text));
}

void loop()
{
}`

rweather, 2020-07-01:
You need to reset the key and IV back to the original state after calling encryptData() and before decryptData().  Otherwise it will try to decrypt with the next IV after the encrypted packet rather than decrypting with the original IV from the start.

franciscone14, 2020-07-03:
Ohh I see. Thank you so much. 
I'm relatively new to Cryptography and just getting the hang of it !

Problem Fixed.

closed 2020-07-03 by franciscone14