rweather/arduinolibs #59: Hashing Problem

RicardoPires19, 2020-06-30:
There seems to be a problem with the hashing method when reading from a file. In this case I'm using SHA256.
When I open a file and read its contents and update the hash, the resulting value is wrong.

I don't understand what I'm doing wrong or if its a bug, but I can't get the correct value by reading a file and updating the hash.
This is my function:

```
bool calculate_file_hash(char* path, byte* result) {
  File f = SD.open(path, FILE_READ);
  if (!f) {
    return false;
  }

  SHA256 hash;
  hash.clear();
  hash.reset();
  
  while(f.available()){
    hash.update(f.read(), 1);
    if(f.peek() < 0)
      break;
  }
  f.close();
  hash.finalize(result, 32);

  return true;
}
```

It reads 1 byte and updates the hash. I've also tried reading chunks of 128 bytes, but it produces a different incorrect hash. When I transfer a file over serial and update the hash with successive 128 bytes, the final hash is correct. For some reason, reading a file with the SD.h library just doesn't work. In fact simply changing the filename from "t.txt" to "/t.txt" produces a different hash.
Am I doing anything wrong?

rweather, 2020-07-01:
I believe that f.read() reads and returns a single character.  But the hash.update() function expects a pointer as its argument.  So it is turning the character into a pointer and reading whatever rubbish was at that position in memory.  You should probably do something like this instead:

char ch = f.read();
hash.update(&ch, 1);

RicardoPires19, 2020-07-01:
Thanks, that fixed the problem. I'll be closing the issue now.

closed 2020-07-01 by RicardoPires19