rweather/arduinolibs #63: Use of htobe64 and other non-standard functions

Anton-V-K, 2020-10-13:
The functions like `htobe64` (it is used in `SHA512::finalize`) are considered non-standard ([ref](https://linux.die.net/man/3/htobe64)), so their usage makes it hard to build the library for other platforms.
It would be nice to replace them with standard functions to make the library easily portable to other platforms.

PS: I was trying to build the library with Microsoft C++ compiler (VS2017)

rweather, 2020-10-13:
The intention was that platform-specific or compiler-specific alternatives would be provided via #ifdef's in EndianUtil.h, as it is hard to do endian conversions efficiently in standard C in a manner that is safe against side-effects.

The byteswap instrinsics in Visual C might do what you want:
https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/byteswap-uint64-byteswap-ulong-byteswap-ushort?view=vs-2019

If you can get it to work, then submit a merge request and I'll add it to the library.

Anton-V-K, 2020-10-14:
Thanks for the reply.
I'm not familiar with gcc-specific keyword `__extension__` and could be overlooking something, but I'm wondering why functions like `htobe64` was written as C-macros and not as inline functions?
For instance, the existing code
```
#define htobe32(x)  \
    (__extension__ ({ \
        uint32_t _temp = (x); \
        ((_temp >> 24) & 0x000000FF) | \
        ((_temp >>  8) & 0x0000FF00) | \
        ((_temp <<  8) & 0x00FF0000) | \
        ((_temp << 24) & 0xFF000000); \
    }))
```
can be easily transformed into portable inline function like:
```
inline uint32_t htobe32(uint32_t x)  {
        uint32_t _temp = x;
    return
        ((_temp >> 24) & 0x000000FF) |
        ((_temp >>  8) & 0x0000FF00) |
        ((_temp <<  8) & 0x00FF0000) |
        ((_temp << 24) & 0xFF000000);
}
```

MauroMombelli, 2020-10-27:
there seems to be huge difference in the way the code is compiler, the version with extension looks like more than double long, see https://godbolt.org/z/jfzK9G