rweather/arduinolibs #65: Simple encrypt/decrypt with public-key algorithms

sam-k, 2021-04-08:
It looks like the public-key algorithms are implemented for very limited use cases: Curve25519 for Diffie–Hellman only, and Ed25519 for digital signatures only.

Is there a way to use asymmetric encryption/decryption on data directly? I'm interested in a digital envelope where the two parties already know the public and private keys.

rweather, 2021-04-08:
On Thu, Apr 8, 2021 at 8:30 PM Sam Kim ***@***.***> wrote:

> It looks like the public-key algorithms are implemented for very limited
> use cases: Curve25519 for Diffie–Hellman only, and Ed25519 for digital
> signatures only.
>
> Is there a way to use asymmetric [...]
It is possible to use some asymmetric schemes like RSA to do direct data
encryption, albeit very slowly.  But unfortunately elliptic curve systems
aren't built to directly operate on user-supplied data (other than the
keys).

Typically you would use the shared secret from a Curve25519 exchange to
generate an encryption key for AES or another traditional cipher, and then
encrypt with that key and cipher.  To authenticate the exchange, hash the
encrypted message and sign the hash with Ed25519.

Cheers,

Rhys.

sam-k, 2021-04-08:
I see. Unfortunately, a DH exchange isn't practical in my particular case, since all communication is assumed to be one-way only. But I didn't know about the limitation of elliptic curves that you mentioned. Thank you very much for your help.

closed 2021-04-08 by sam-k

rweather, 2021-04-08:
The Noise protocol is a framework for constructing secure communications methods using elliptic curve cryptography: http://noiseprotocol.org/noise.html

Noise has support for one-way patterns that can be used to encrypt a packet to another party without having an online connection, as long as you know the public key of the other party ahead of time: http://noiseprotocol.org/noise.html#one-way-handshake-patterns

Noise may seem a bit complex, but it is actually quite simple compared to other approaches like PKCS#7 and TLS.  Noise is used in the in-kernel WireGuard VPN protocol (in interactive mode), so there is some real world use of the framework out there.