rweather/arduinolibs #66: Error compiling for new Arduino core for the ESP32

TiagoPereiraRibeiro, 2021-09-09:
"hwcrypto/aes.h: No such file or directory"

Board selected on Arduino IDE: ESP32 Dev Module

Comments: 

- It appears that with Arduino core for the ESP32 version 2.0, the folder hwcrypto no longer exists.
   - Compile correctly with the previous version (1.0.6)

garudaonekh, 2021-11-12:
Same issue, any suggestion?

PiotrBzdrega, 2021-11-19:
Guy below mentioned  that _**hwcrypto/***_ has been moved to _**esp32/***_.
[https://githubmemory.com/repo/kind3r/esp32-ble-gateway/issues/3](url)

So i the file \Documents\Arduino\libraries\Crypto\src\AES.h:200: i changed
`#include "hwcrypto/aes.h"` ->   `#include "esp32/aes.h"`  then i was able to compile prj but.... with warnings
#warning "esp32/aes.h is deprecated, please use aes/esp_aes.h instead"

After change for `#include "aes/esp_aes.h"` problem disappeared.

Idk if this commit is related to this somehow :
[https://gitea.edwinclement08.com/espressif/esp-idf/commit/709a320f337572f0ad6fb9c6b2d162bd3005cf93](url)

spleenware, 2021-12-28:
I'm also encountering this. Any way to upvote this issue?

rweather, 2021-12-30:
I have attempted to fix this in the latest version by defining the ESP32 AES library functions myself rather than include a system header that keeps moving around in different SDK versions.  Hopefully this helps resolve the problem permanently.  If not, let me know.

closed 2021-12-30 by rweather