rweather/arduinolibs #75: Bug of Ascon128.c++ in Arduino Cryptography Library

Oulalaa, 2023-02-22:
Hi, Rhys!

First, thank you so much for your contribution to the Arduino Cryptography Library! 

I met an error when I tried to use the Ascon128 algorithm in the library to test performance on an Arduino Mega. I believe it is the variable type that caused the error. The error is as follows.

```
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o: In function `Ascon128::endAuth()':
/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:297: undefined reference to `Ascon128::permute(unsigned char)'
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o: In function `Ascon128::checkTag(void const*, unsigned int)':
/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:225: undefined reference to `Ascon128::permute(unsigned char)'
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o: In function `Ascon128::addAuthData(void const*, unsigned int)':
/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:176: undefined reference to `Ascon128::permute(unsigned char)'
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o: In function `Ascon128::decrypt(unsigned char*, unsigned char const*, unsigned int)':
/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:149: undefined reference to `Ascon128::permute(unsigned char)'
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o: In function `Ascon128::encrypt(unsigned char*, unsigned char const*, unsigned int)':
/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:120: undefined reference to `Ascon128::permute(unsigned char)'
/var/folders/f2/gzz10k7506x8jq2cmwxgjl140000gn/T//ccIMWpI8.ltrans0.ltrans.o:/Users/achiless/Documents/Arduino/libraries/Crypto/src/Ascon128.cpp:199: more undefined references to `Ascon128::permute(unsigned char)' follow
collect2: error: ld returned 1 exit status
exit status 1
```

It looks like something goes wrong with the variable type of function permute().  My Arduino code is as follows.

```
#include <Ascon128.h>
#include "Crypto.h"
#include "utility/EndianUtil.h"
#include "utility/RotateUtil.h"
#include "utility/ProgMemUtil.h"
#include <string.h>
 
void setup() {
Serial.begin(9600);
}
void loop() {
Ascon128 test;
}
```

Then I changed the permute(6) in line 199 in Ascon128.c++ into permute((uint8_t)6). But the error was still there.

I could not why is it and how to figure it out. Could you help to fix this bug and if possible update a new version? It is also OK if you have no time, I completely understand. I am looking forward to your reply.

rweather, 2023-02-22:
The error is referring to Crypto/src/Ascon128.cpp, but the ASCON code is usually in the CryptoLW library.

Are you using the latest version of the Crypto and CryptoLW libraries from this repository?

https://github.com/rweather/arduinolibs

Sometimes third-party people push versions of my library up to Arduino package repositories but I don't maintain those.  They can be very out of date.  I suggest uninstalling anything from a package repository and use the main copy instead.

Anyway, you will probably need Ascon128AVR.cpp to provide the AVR implementation of Ascon128::permute().  That may be the thing that is missing.

If you want to use ASCON specifically instead of all the other algorithms, then I have a new repository with more recent and faster ASCON implementations here:

https://github.com/rweather/ascon-suite

Oulalaa, 2023-02-22:
Thank you so much for your quick reply and kind tips!

I downloaded the lib from Arduino IDE, and added Ascon128.h and .cpp from
the website. I will do as your suggestions to replace the old Arduino
package repositories.
I am planning to use Ascon in my next paper. Thank you again for your
kindly help! I will refer to your lib and acknowledge you in my paper if it
can be published.

Best,
Chao

rweather, 2023-02-22:
Hi Chao.  Let me know if you have any further issues and good luck with your paper.

closed 2023-02-22 by rweather