rweather/arduinolibs #77: Crypto doesn't support next generation AVRs

specternecter, 2023-03-17:
Since Atmel was taken over by Microchip, a lot has changed with AVRs. Timers are completely different now and when I try to run this library with an ATTINY3226 it gives tons of compilation errors because it's referencing timers that don't exist. Also, watchdog timers are not recommended according to the core developer for megatinycore. Would it be possible to port new AVRs such as the ATTINY3226? I wouldn't know where to begin.

specternecter, 2023-09-04:
It shouldn't give build errors for the ATTiny3224 as I integrated the new ATTinys into the library shortly after I raised this issue. There has been an update or two to the IDE since then though. What errors are you getting with each chip and what version of the IDE are you using?

specternecter, 2023-09-04:
Actually, I guess more importantly... what version of megaTinyCore are you using? The newer versions should include the fix unless Spence changed something that interferes with it because it was actually his idea to use CCL for the RNG clock and implemented it. That being said, if you're using CCL for other purposes it will never work with crypto together.

FerGT50, 2023-09-04:
I am using the latest MegaTinyCore 2.6.8 and Arduino 1.8.19.

Here's the log (example "TestEd25519"):

````
Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204, ATtiny3224, 20 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), PB0-2, PA3-5 (default), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

In file included from C:\Users\Fernando\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8\cores\megatinycore/Arduino.h:37:0,

                 from L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:27:

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp: In function 'void WDT_vect()':

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:228:5: warning: 'WDT_vect' appears to be a misspelled 'signal' handler, missing '__vector' prefix [-Wmisspelled-isr]

 ISR(WDT_vect)

     ^

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:239:27: error: 'TCNT0' was not declared in this scope

     unsigned char value = TCNT0;

                           ^~~~~

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:239:27: note: suggested alternative: 'TCA0'

     unsigned char value = TCNT0;

                           ^~~~~

                           TCA0

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp: In destructor 'RNGClass::~RNGClass()':

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:288:5: error: 'MCUSR' was not declared in this scope

     MCUSR &= ~(1 << WDRF);

     ^~~~~

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:288:21: error: 'WDRF' was not declared in this scope

     MCUSR &= ~(1 << WDRF);

                     ^~~~

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:288:21: note: suggested alternative: 'WDT'

     MCUSR &= ~(1 << WDRF);

                     ^~~~

                     WDT

In file included from L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:37:0:

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:291:30: error: 'WDCE' was not declared in this scope

     _WD_CONTROL_REG |= (1 << _WD_CHANGE_BIT) | (1 << WDE);

                              ^

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:291:30: note: suggested alternative: 'ADC0'

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:291:54: error: 'WDE' was not declared in this scope

     _WD_CONTROL_REG |= (1 << _WD_CHANGE_BIT) | (1 << WDE);

                                                      ^~~
[... 85 lines cut ...]
     _WD_CONTROL_REG |= (1 << _WD_CHANGE_BIT) | (1 << WDE);

                                                      ^~~

                                                      WDT

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:490:29: error: 'WDIE' was not declared in this scope

     _WD_CONTROL_REG = (1 << WDIE);

                             ^~~~

L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:490:29: note: suggested alternative: 'WDT'

     _WD_CONTROL_REG = (1 << WDIE);

                             ^~~~

                             WDT

exit status 1

Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.

````

FerGT50, 2023-09-04:
Comment from Spence Konde:

`it can only be dealt with at the library level as the library needs to use a different means to gather entropy (on classic they used the beat of the poorly controlled, asynchronous WDT oscillator against the system clock to generate entropy; on the new parts, there turns out to be a better way that generates more entropy faster (though an analogous method using the RTC is also possible, but in that thread, I got the impression that my CCL hack was faster and - besides that I suspect it's a better source of randomness, since Microchip isn't even trying control the timing of a CCL inverting itself`

specternecter, 2023-09-04:
Ok, what version of THIS library are you using? Because your first error is "L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp:228:5: warning: 'WDT_vect' appears to be a misspelled 'signal' handler, missing '__vector' prefix [-Wmisspelled-isr]". In the new version of RNG.cpp line 228 is a comment, not an instruction. If you're running the latest version of this library, you shouldn't be able to get that error. But if for some reason something is broken, replace the file in your directory at L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp with the file at https://github.com/rweather/arduinolibs/blob/master/libraries/Crypto/RNG.cpp

FerGT50, 2023-09-04:
I am using 0.4.0

specternecter, 2023-09-04:
I don't understand exactly what's going on here but I really don't think you have the proper RNG.cpp file. There are different forks of Crypto and I'm wondering if maybe you're using a different one? The last change to this library that I see is when the new RNG.cpp file I changed got added in April of 2023. Regardless though, your error shouldn't come up if you have the new RNG.cpp file. WDT_vect doesn't exist in the new AVRs and the new RNG.cpp uses CCL instead. The new RNG.cpp tries to determine if the device is a new AVR with CCL (yours is) and uses that instead of WDT. If you replace the file manually, it should work because that's the only file that needed changed to enable use with modern AVRs like the one you're using.

specternecter, 2023-09-04:
If you take a look at your file at L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp I can almost guarantee line 38 doesn't say `#if defined(MEGATINYCORE) || defined(LOGIC_H) // If device is Modern AVR with CCL` as it should

specternecter, 2023-09-04:
Yeah I looked at it and your path should look more like this if you're using this library `"L:\Progetti\Arduino\libraries\arduinolibs-master\libraries\Crypto\RNG.cpp"` . You should remove the Crypto library you have and manually replace it with this one

specternecter, 2023-09-04:
My path looks like this `C:\Users\brad\Downloads\arduinolibs-master\libraries\Crypto\RNG.cpp`  and yours looks like this `L:\Progetti\Arduino\libraries\Crypto\src\RNG.cpp`. Notice yours has a `\src\` folder that isn't present in this library. You should consider manually replacing your whole Crypto library with this one

FerGT50, 2023-09-04:
I installed the Crypto library from the Arduino IDE Library Manager.
It is the latest one, 0.4.0.
There is not a newer version to be installed.

specternecter, 2023-09-04:
Yeah, it looks like the Arduino IDE Library Manager is referencing a different fork. Do you know how to remove it and install this one instead or do you want me to walk you through it?

FerGT50, 2023-09-04:
No, thanks, I can manage it: I just wanted to be sure the "official" one from the Library Manager was not the right one.
Of course this means a lot of people is going to have this very problem. ':)

specternecter, 2023-09-04:
Yeah unfortunately it does and I don't really know how to go about changing it aside from figuring out which fork they're using and suggesting they replace that RNG.cpp file

FerGT50, 2023-09-04:
Ok, replaced all files with the ones from the latest github zip from master branch, now TestEd25519 builds fine.
Thanks!

FerGT50, 2023-09-04:
PS: maybe the fact you still use "0.4.0" as library version may lead to issues with Arduino IDE Library Manager (at least with 1.8.x)?
BTW IDE 2.x is unbearable... extremely, painfully slow despite 32 GB RAM and SSD.

specternecter, 2023-09-04:
No problem. I did a little searching and believe I found the library you had. It's at https://github.com/OperatorFoundation/Crypto and turns out it's a fork of THIS library but without the proper RNG.cpp file lol. I'll submit a new issue and suggest they change it. People will probably appreciate you running into this problem

specternecter, 2023-09-04:
Well, it turns out I can't submit a new issue and whenever I try to do a pull request it ends up bringing me back to this library (admittedly I don't have any experience with pull requests lol). Hopefully they fix it soon or people at least find this issue and realize how to fix it.

specternecter, 2023-09-04:
Actually, the more I look into it, the problem appears to trace back to this library but it's really strange because if I follow the links from the Library Manager to the RNG.cpp file, it is the current one. I don't know what's going on but it's outside my expertise.