rweather/arduinolibs #78: ’SHA256‘  was not declared in this scope ESP32S3

KINGJ7, 2023-03-28:
Hi, I've been using #include<SHA256.h> #<Crypto.h> on Arduino ide2.0.4 with my ESP32S3,
 Error compiling for board ESP32S3（2.0.7）version

【/Users/king.j1/Documents/Arduino/libraries/YYZT_AliyunMqtt/src/YYZT_AliyunMqtt.cpp: In member function 'String AliyunMqtt::hmac256(const String&, const String&)':
/Users/king.j1/Documents/Arduino/libraries/YYZT_AliyunMqtt/src/YYZT_AliyunMqtt.cpp:295:3: error: 'sha256' was not declared in this scope
   sha256.resetHMAC(key, keySize);
   ^~~~~~
/Users/king.j1/Documents/Arduino/libraries/YYZT_AliyunMqtt/src/YYZT_AliyunMqtt.cpp:295:3: note: suggested alternative: 'hmac256'
   sha256.resetHMAC(key, keySize);
   ^~~~~~
   hmac256】

How to solve this problem？Thanks

rweather, 2023-03-29:
I suspect that there is a conflict with the headers in the toolchain for the ESP32S3.  Maybe it has its own version of Crypto.h or SHA256.h and that header is getting picked up instead.  It wouldn't be the first time that the system headers for ESP32 have conflicted in this way.  I'm not really sure how to force a sketch to use the headers in my library instead of the system headers.  You may want to also talk to the authors of AliyunMqtt to see if they have a solution.

KINGJ7, 2023-03-29:
> I suspect that there is a conflict with the headers in the toolchain for the ESP32S3. Maybe it has its own version of Crypto.h or SHA256.h and that header is getting picked up instead. It wouldn't [...]

I also get this error using the SHA256.io and the header of SHA256.h ， in your Crypto library‘s example

KINGJ7, 2023-03-30:
> I suspect that there is a conflict with the headers in the toolchain for the ESP32S3. Maybe it has its own version of Crypto.h or SHA256.h and that header is getting picked up instead. It wouldn't [...]

I have no problem using other libraries, such as SHA224.h and SHA384.h and so on，The problem seems to be unique to the SHA256.h ，If you know how to solve it, please let me know，thanks

rweather, 2023-03-31:
It looks like your ESP toolchain has both Crypto.h and SHA256.h headers, which are conflicting with the ones in my library.  Here are some options:

- Rename the files in my library to something else like CryptoA.h and SHA256A.h or something.  And then change anything that refers to them.
- Rename the headers in the toolchain itself.
- Find some way to modify the Arduino IDE's or the toolchain's include path to put my library first in the path.

Sorry I cannot be much more help than that.

KINGJ7, 2023-04-01:
> It looks like your ESP toolchain has both Crypto.h and SHA256.h headers, which are conflicting with the ones in my library. Here are some options:
>
> * Rename the files in my library to something else [...]

Thanks bro，I have solved the problem^_^

rweather, 2023-04-01:
For my curiosity and the benefit of others with the same problem, how did you end up solving the problem?

closed 2023-04-02 by rweather