rweather/arduinolibs #82: crypto_feed_watchdog was not declared in this scope [ESP8266 & ESP32]

alve89, 2023-10-03:
I just tried to upload the Speck example to both an ESP8266 and an ESP32 but the compilation ended with the following error:

> In function 'void perfCipher(BlockCipher*, const TestVector*, size_t, bool)':
> C:\Users\xxxx\AppData\Local\Temp\.arduinoIDE-unsaved202393-20620-1c3apjw.9qoy\TestSpeck\TestSpeck.ino:110:5: error: 'crypto_feed_watchdog' was not declared in this scope

So it's the following line which causes this problem: [CryptoLW/examples/TestSpeck/TestSpeck.ino => Line 81](https://github.com/rweather/arduinolibs/blob/1e13ef7b244caaa1b9151d5e41c4cdb029f4ea84/libraries/CryptoLW/examples/TestSpeck/TestSpeck.ino#L81C7-L81C7)

Using this example on an Arduino Nano works properly.

How can tihs issue be solved?

Many thanks in advance!

rweather, 2023-10-03:
Recent ESP SDK's have their own Crypto.h that conflicts with mine.  It wasn't always like this - they have recently taken over a lot of generic header file names in the SDK rather than use safe names like EspCrypto.h or something.  Arduino Nano doesn't have this problem because that SDK doesn't override Crypto.h.  The sketch is probably picking up the ESP SDK's Crypto.h file.  You will need to either rename the SDK's file or mine to work around this.

alve89, 2023-10-03:
Thank you for this very quick reply!

I followed your suggestion and

1. renamed
`./libraries/Crypto/Crypto.h` to `./libraries/CryptoCstm/CryptoCstm.h`
`./libraries/Crypto/Crypto.cpp` to `./libraries/CryptoCstm/CryptoCstm.cpp`
2. changed the line
`#include <Crypto.h` to `#include <CryptoCstm.h>`

This ended in the following:

> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp: In destructor 'virtual ChaCha::~ChaCha()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp:55:5: error: 'clean' was not declared in this scope
>      clean(block);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp:55:5: note: suggested alternative: 'clear'
>      clean(block);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp: In member function 'virtual void ChaCha::clear()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp:220:5: error: 'clean' was not declared in this scope
>      clean(block);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaCha.cpp:220:5: note: suggested alternative: 'clear'
>      clean(block);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp: In destructor 'virtual AESCommon::~AESCommon()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp:48:5: error: 'clean' was not declared in this scope
>      clean(ctx, sizeof(ctx));
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp:48:5: note: suggested alternative: 'clear'
>      clean(ctx, sizeof(ctx));
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp: In member function 'virtual void AESCommon::clear()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp:83:5: error: 'clean' was not declared in this scope
>      clean(ctx, sizeof(ctx));
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\AESEsp32.cpp:83:5: note: suggested alternative: 'clear'
>      clean(ctx, sizeof(ctx));
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp: In destructor 'virtual BLAKE2b::~BLAKE2b()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:96:5: error: 'clean' was not declared in this scope
>      clean(state);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:96:5: note: suggested alternative: 'clear'
>      clean(state);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp: In member function 'virtual void BLAKE2b::clear()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:245:5: error: 'clean' was not declared in this scope
>      clean(state);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:245:5: note: suggested alternative: 'clear'
>      clean(state);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp: In member function 'virtual void BLAKE2b::finalizeHMAC(const void*, size_t, void*, size_t)':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:265:5: error: 'clean' was not declared in this scope
>      clean(temp);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\BLAKE2b.cpp:265:5: note: suggested alternative: 'clear'
>      clean(temp);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\CTR.cpp: In destructor 'virtual CTRCommon::~CTRCommon()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\CTR.cpp:53:5: error: 'clean' was not declared in this scope
[... 66 lines cut ...]
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp:159:18: note: suggested alternative: 'secure_getenv'
>      bool equal = secure_compare(temp, tag, len);
>                   ^~~~~~~~~~~~~~
>                   secure_getenv
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp:160:5: error: 'clean' was not declared in this scope
>      clean(temp);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp:160:5: note: suggested alternative: 'clear'
>      clean(temp);
>      ^~~~~
>      clear
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp: In member function 'virtual void ChaChaPoly::clear()':
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp:168:5: error: 'clean' was not declared in this scope
>      clean(state);
>      ^~~~~
> c:\Users\xxxxx\xxxxx\Privat\Projekte\Arduino\libraries\CryptoCstm\ChaChaPoly.cpp:168:5: note: suggested alternative: 'clear'
>      clean(state);
>      ^~~~~
>      clear
> 
> exit status 1
> 
> Compilation error: exit status 1

>

alve89, 2023-10-03:
I solved it by changing all `#include` statements.

closed 2023-10-03 by alve89