rweather/arduinolibs #84: RNG.cpp include of <Arduino.h> makes native builds for unit testing and debugging fail

attermann, 2023-11-21:
RNG.cpp include <Arduoino.h> and makes use of Arduino millis() and micros() functions.

It's evident from code in the repo that this issue was circumvented for this project by emulating Arduino (.../host/emulation/Arduino.h), but it's not clear how this emulation can be included by external projects.

I'm using PlatformIO specifically. Any advice for how to include the emulation for a native host build?

Thanks!

jp-bennett, 2024-08-09:
You may be interested in platform-native. https://docs.platformio.org/en/latest/platforms/native.html It's been a long time since a stable release, but the code itself is still under development at https://github.com/meshtastic/platform-native

Adrian-Samoticha, 2024-08-23:
> You may be interested in platform-native. https://docs.platformio.org/en/latest/platforms/native.html It's been a long time since a stable release, but the code itself is still under development at https://github.com/meshtastic/platform-native

Unless I’m misunderstanding something, PlatformIO’s platform-native functionality is exactly what @attermann is using for unit testing purposes, and the inclusion of `Arduino.h` makes it impossible to compile this library on native environments.

I‘ve run into the same issue and am currently looking for a solution.

Adrian-Samoticha, 2024-08-30:
I couldn’t really find a good solution to this other than replacing the `#include<Arduino.h>` line in the `RNG.cpp` file with:

```cpp
#ifndef NATIVE
#include <Arduino.h>
#else
uint32_t millis() { return 0; }

uint32_t micros() { return 0; }
#endif
```